sparkfun/SparkFun_TMAG5273_Arduino_Library #2: Inconsistent readings with "Basic Readings" example

Raelx, 2023-09-19:
Running the "Basic Readings" example with a stationary magnet near the sensor produces very inconsistent values. Serial output values seem to bounce between near zero and what might be the actual mT values (in the range of 10-50mT). Here is some example output.

28.39 C
(0.16, -15.59, 36.52) mT
28.39 C
(0.16, -0.08, -0.31) mT
28.13 C
(-60.74, -0.08, 36.52) mT
28.66 C
(-60.90, -0.08, -0.27) mT
28.13 C
(0.12, -0.08, -0.27) mT
28.39 C
(0.12, -0.04, 36.45) mT
28.39 C
(0.16, -0.04, 36.45) mT
28.13 C
(0.20, -0.08, -0.27) mT
28.39 C
(0.20, -15.00, -0.27) mT
28.66 C
(0.23, -15.00, 36.52) mT
27.86 C
(0.08, -0.12, -0.31) mT
28.66 C

### Your workbench
* What development board or microcontroller are you using?
  * ESP8266 (Wemos D1 mini)
* What version of hardware or breakout board are you using? 
  * Mikroe 3D Hall 11 Click
* How is the breakout board wired to your microcontroller? 
  * 3.3v power
  * 10k pull up resistors on SCL and SDA
* How is everything being powered?
  * USB
* Are there any additional details that may help us help you?
  * Had to specify the I2C address for the sensor as 0x35, this Library defaults to 0x22

### Steps to reproduce
Just run the example code

### Expected behavior
Read values should be more or less stable without dropping out to near zero for most reading.

### Actual behavior
Read value fluctuate between zeroish and a believable value.

Raelx, 2023-09-21:
Seems like this is due to some incorrect math combining the LSB and MSB for the mag values. I have submitted a pull request to fix this. Working fork is [available here](https://github.com/Raelx/SparkFun_TMAG5273_Arduino_Library) in the meantime.

MadisonC-SparkFun, 2023-09-26:
See latest PR for more information

closed 2023-09-26 by MadisonC-SparkFun

asfarley, 2023-12-01:
Hi, I see this issue has been closed but I'm still running into the same thing. Is this expected to be fixed in 1.0.3? 

I'm testing the BasicReadings example and seeing essentially the same results as @Raelx except that the Z-axis readings appear to be stable. The X and Y axis readings are still jumping around different values in a way that looks more like conversion error than noise.

asfarley, 2023-12-01:
I'm also seeing -266 Celsius for an indoor temperature reading so maybe I'm having a different issue.

MadisonC-SparkFun, 2023-12-01:
Hi @asfarley, this magnetic axis reading issue will be fixed in the 1.0.3 release - that was released 2 days ago so it should be in the Arduino Library Manager ready to use now. 
As for the temperature reading, I have not seen this issue personally. If you would like, you can share your code and environment so I can look into it.