sparkfun/SparkFun_TMAG5273_Arduino_Library #10: Doesn't support multiple I2C ports

goodmanj, 2025-01-13:
The TMAG5273 class includes an `_i2cPort` field that gets initialized by the `begin()` method, so it *should* allow you to talk to TMAG sensors on multiple I2C ports on microcontrollers that support that.  But the implementation of `readRegisters() `and `writeRegisters()` explicitly use the `Wire` object, so they can't be used for `Wire1` etc.

I tried forking the repo to fix this myself but ran into technical difficulties, not sure why.

Segror, 2025-04-02:
Thanks for the hint, i was missing that.

You can easily fix it by replacing all "Wire." (with the dot) by "_i2cPort->" (with the arrow)
you need to change the dot to the arrow since i2cPort is a Pointer.

BTW its dangoerous to use TMAG5273_I2C_ADDRESS_INITIAL since according to the datasheet it depends on the exact Chip version, there are 4 possible initial i2c addresses:

see: Table 6-2 on page 16 of the datasheet:
https://www.ti.com/lit/ds/symlink/tmag5273.pdf

("read adress" and "write adress" can be ignored, its just the leftshifted i2c Address, once with 1 as LSB, once with a 0)

gigapod, 2025-12-12:
This was fixed with version 2.0 of the library. You can now pass in a Wire Port that uses the desired pins.