sparkfun/SparkFun_TMAG5273_Arduino_Library #11: Bug in `readRegister()` – Incorrect register byte count in `readRegisters` call

justkozi, 2025-05-09:
# Bug in `readRegister()` – Incorrect register byte count in `readRegisters` call

## Description

While working with the **TMAG5273** sensor on the **Raspberry Pi Pico** using your library, I encountered a communication issue which I traced back to the `readRegister()` function.

### Problem

The current implementation reads **2 bytes** from a register instead of the expected **1 byte**, which leads to incorrect behavior on devices with strict I2C read handling like the Pi Pico.

```cpp
uint8_t TMAG5273::readRegister(uint8_t regAddress)
{
    uint8_t regVal = 0;
    readRegisters(regAddress, &regVal, 2);  // <-- Incorrect byte count
    return regVal;
}
```

### Fix

```cpp
uint8_t TMAG5273::readRegister(uint8_t regAddress)
{
    uint8_t regVal = 0;
    readRegisters(regAddress, &regVal, 1);  // <-- Should read only 1 byte
    return regVal;
}
```

## Impact

- Causes data corruption or bus errors on microcontrollers with strict I2C behavior
- Verified specifically on Raspberry Pi Pico (RP2040)

Please consider updating the library accordingly. Let me know if a pull request is preferred—I’d be happy to submit one.

kdyke, 2025-07-13:
I just hit this as well trying to use this library with the SparkFun SAMD21 board.    Among other things this also trashes values on the stack, causing functions that call this to wind up with locally corrupted values, and this causes the sensor init to fail for me.   The read of 2 bytes is very clearly wrong as it’s trying to read two bytes where there is only storage for one.

thebluewizards, 2025-12-04:
I agree. This fixed some of my issues when trying to wire multiple boards simultaneously